asukiaaa/gerber_to_order #4: Error: 'PCB_TARGET' object has no attribute 'GetShapeStr'

exislow, 2021-11-05:
I have installed the plugin with KiCAD 5.1.10-1 on my macOS 11.6. If I try to click on `export` using this plugin the following error appears:

```
Error: 'PCB_TARGET' object has no attribute 'GetShapeStr'

Traceback (most recent call last):
File "/Users/xxx/Library/Preferences/kicad/scripting/plugins/gerber_to_order/gerber_to_order_action.py", line 327, in OnExec
sizeLabel = createSizeLabelOfBoard(pcbnew.GetBoard())
File "/Users/xxx/Library/Preferences/kicad/scripting/plugins/gerber_to_order/outline_measure.py", line 106, in createSizeLabelOfBoard
wh = getWidthHeightMmOfBoard(board)
File "/Users/xxx/Library/Preferences/kicad/scripting/plugins/gerber_to_order/outline_measure.py", line 99, in getWidthHeightMmOfBoard
minMax2Dim = getMinMax2DimOfBoard(board)
File "/Users/xxx/Library/Preferences/kicad/scripting/plugins/gerber_to_order/outline_measure.py", line 71, in getMinMax2DimOfBoard
if draw.GetShapeStr() == 'Arc':
AttributeError: 'PCB_TARGET' object has no attribute 'GetShapeStr'
``` 

This is my PCB: 
[image]

What have I done wrong?

asukiaaa, 2021-12-06:
Thank you for sharing the problem.
Can you share the project?
I want to reproduce the problem on my PC.

odaki, 2021-12-06:
Hello @exislow.
This problem has been fixed by #3 .
Please try the latest version.

exislow, 2021-12-10:
Thanks for your effort. Currently I am on KiCad 5.99 (nightlies). After your patch I get the follwing error message:

```
Error: module 'pcbnew' has no attribute 'DRAWSEGMENT'

Traceback (most recent call last):
File "/Users/USER/Documents/KiCad/6.0/scripting/plugins/gerber_to_order/gerber_to_order_action.py", line 327, in OnExec
sizeLabel = createSizeLabelOfBoard(pcbnew.GetBoard())
File "/Users/USER/Documents/KiCad/6.0/scripting/plugins/gerber_to_order/outline_measure.py", line 106, in createSizeLabelOfBoard
wh = getWidthHeightMmOfBoard(board)
File "/Users/USER/Documents/KiCad/6.0/scripting/plugins/gerber_to_order/outline_measure.py", line 99, in getWidthHeightMmOfBoard
minMax2Dim = getMinMax2DimOfBoard(board)
File "/Users/USER/Documents/KiCad/6.0/scripting/plugins/gerber_to_order/outline_measure.py", line 70, in getMinMax2DimOfBoard
if type(draw) is pcbnew.DRAWSEGMENT and draw.GetLayerName() == 'Edge.Cuts':
AttributeError: module 'pcbnew' has no attribute 'DRAWSEGMENT'

```

exislow, 2021-12-10:
I have tried the docker version: Works like charm.