asukiaaa/gerber_to_order #5: kicad6

shachooo, 2021-12-27:
便利なので使わせてもらっています
kicad6リリースしたので、試してみたところ、エラーが出ました
apiの変更の為、いくつかエラーが出ます
エラー内容は省きますが
以下のようにしたところ、エラーが出なくなりました
問題ないでしょうか？
outline_measure.py

[image]

gerber_to_order_action.py
[image]

[image]

よろしくお願いいたします

asukiaaa, 2021-12-27:
不具合情報ありがとうございます。
pull requestを作成していただけますか？

odaki, 2021-12-28:
ブランチを切ってKiCad6対応してみました。
ただし、現状、KiCad5とは互換性がなくなる書き方なので pull request はしておりません。
https://github.com/odaki/gerber_to_order/tree/kicad6

odaki, 2021-12-28:
https://github.com/yaqwsx/pcbnewTransition
のisV6()のようなものを用意して、5.xと6.xの両対応するのがベターかと思いつつも、そこまでできておりません。

shachooo, 2021-12-28:
ありがとうございます
使ってみます

odaki, 2021-12-29:
@asukiaaa @shachooo KiCad 6 (5.99～)と、KiCad 5の両方に対応できるように修正してみました。
よろしければマージをお願いします。
Support both KiCad 5 and KiCad 6 https://github.com/asukiaaa/gerber_to_order/pull/6

asukiaaa, 2021-12-29:
ご対応ありがとうございます。
マージしました。

odaki, 2021-12-29:
マージいただきどうもありがとうございました!
issue #4 の対策にもなっているはずです。

closed 2022-01-05 by shachooo