asukiaaa/gerber_to_order #8: How to install on Kicad 7.0.1 ?

Witawat, 2023-04-11:
How to install on Kicad 7.0.1 ?

asukiaaa, 2023-04-11:
Merge request for KiCad Addon is pending so please install as development way.
https://github.com/asukiaaa/gerber_to_order#development
https://gitlab.com/kicad/addons/metadata/-/merge_requests/205

asukiaaa, 2023-04-21:
This plugin can be installed with using GUI of KiCad plugin mnager.

Tools -> Plugin and Content Manager
[image]

Search `gerber to order` then click `Install` and `Apply Changes`.
[image]

closed 2025-01-05 by Witawat