asukiaaa/gerber_to_order #9: error thrown on export

fbd-ss, 2023-08-23:
Hi please see the following, can supply pcb if you want

[image]

Cheers

asukiaaa, 2023-08-23:
Can you try uninstalling and installing this plugin?
If the problem continue after installing again, I want you to share a kicad_pcb file that calls this error.

fbd-ss, 2023-08-23:
no go, heres the pcb, it works on everything else just not this. 

Application: KiCad x64 on x64

Version: 7.0.6, release build

Libraries:
	wxWidgets 3.2.2
	FreeType 2.12.1
	HarfBuzz 6.0.0
	FontConfig 2.14.1
	libcurl/7.88.1-DEV Schannel zlib/1.2.13

Platform: Windows 11 (build 22621), 64-bit edition, 64 bit, Little endian, wxMSW

Build Info:
	Date: Jul  6 2023 04:56:45
	wxWidgets: 3.2.2 (wchar_t,wx containers)
	Boost: 1.81.0
	OCC: 7.6.3
	Curl: 7.88.1-DEV
	ngspice: 40
	Compiler: Visual C++ 1936 without C++ ABI

Build settings:
	KICAD_SPICE=ON

[10_pin_atmel_ice_breakout.zip](https://github.com/asukiaaa/gerber_to_order/files/12423601/10_pin_atmel_ice_breakout.zip)

asukiaaa, 2023-08-23:
I could create zip files without any problem.
[image]

My environment
OS: Ubuntu: 22.04
KiCad: 7.0.6

> Can you try uninstalling and installing this plugin?

Did you do this?

fbd-ss, 2023-08-26:
Hey Dean, 
Yes I reinstalled it.  Still the same result.
It was working then all of a sudden its not.

I've had very little to do with python, any idea how I would attach a debugger to the python file when it is running.

xtexChooser, 2023-10-15:
I also reproduced this

asukiaaa, 2023-10-16:
Can you share your project which causes the problem.

xtexChooser, 2023-10-22:
> Can you share your project which causes the problem?

<https://codeberg.org/xtex/micro-carkit.git>

Thanks!