asukiaaa/gerber_to_order #11: Gerber error

Shailesh-khole, 2023-10-20:
I get an error while using it for a board which has a circular cut along the edge.
Unsupported operand type(s) for +: 'float' and 'EDA ANGLE'
outline_measure.py line36 in hasLineOnDegree
angleDegreeEnd = anfleDegreeStart+angleDegree
TypeError:unsupported operand type(s) for +: 'float' and 'EDA ANGLE'
[image]

asukiaaa, 2023-10-21:
Same as https://github.com/asukiaaa/gerber_to_order/issues/9
KiCad sometimes has problem to load old api.
I want you to re install KiCad and plugins.

Shailesh-khole, 2023-10-23:
same error. have updated to the latest version via plugin manager