asukiaaa/gerber_to_order #12: Export error : Error in method 'EXCELLON_WRITE_SetOptions', argument 4 of type 'VECTOR2I const &'

bzhmaddog, 2024-03-13:
It happens when I have 'textures' in the model of one of my footprint

Included below a link to the step file I think is causing this error

https://raw.githubusercontent.com/bzhmaddog/lamp-matrix-tester/main/libs/Models/1088ASBSv5.step

asukiaaa, 2024-03-13:
Can you share files of your project and version of KiCad?

bzhmaddog, 2024-03-14:
Sure thing. I am using Kicad 8 and you can find the project => https://github.com/bzhmaddog/lamp-matrix-tester

bzhmaddog, 2024-03-14:
At some point it was working fine (I believe before I added textures to my model)

asukiaaa, 2024-03-16:
Thank you for sharing the information.
I found a problem about detecting version so I updated it then resolved this problem.
https://github.com/asukiaaa/gerber_to_order/commit/db7a12921de7d971b4d366333e323229dbab3c5f
I created merge request for KiCad Addons so you may can update to version 1.0.3 via addon manager.
https://gitlab.com/kicad/addons/metadata/-/merge_requests/317

bzhmaddog, 2024-03-16:
Great. Thank you.

closed 2025-03-01 by asukiaaa