asukiaaa/gerber_to_order #14: Export failing with error "unsupported operand type(s) for +: 'float' and 'EDA_ANGLE'

francisrath, 2025-02-10:
I have a board with rounded corners using the new KiCad 9 feature that is not exporting.

Steps to reproduce:

Create a new PCB, add a rectangle on `Edge.Cuts` and export using plugin. This works fine.

Close plugin, right click rectangle and click `Shape Modification -> Fillet Lines...` and use a non-zero value, for example 1 mm. Try exporting again and it will fail. Chamfering instead of fillet will work just fine.

[image]

asukiaaa, 2025-02-10:
Thank you for reporting.
I will debug after releasing KiCad9.
Pull request is well come.

asukiaaa, 2025-03-01:
@francisrath 
KiCad9 was released on 2025 Feb 22 so I tried but I couldn't reproduce the problem.
Do you have same problem on released version?
[image]

My environment:
OS: ubuntu22.04
KiCad: 9.0.0